CyberducksTeam/HACKberry #2: [Research] Check prices for the routing of PCBs

Slyde-R, 2023-11-08:


Slyde-R, 2023-11-21:
[image]

Mk2 costs 79€ including delivery - from Hackberry

Slyde-R, 2023-11-29:
soldering the MK2 board is not feasable -> MK1 board with separate arduino

Slyde-R, 2023-11-29:
[image]

[image]

5 of each board (hand, hand_left, sensor, battery) for a total of

Slyde-R, 2023-11-29:
TODO: check footprint and pin out of audio jacks (highlighted in BOM)

weberLeon, 2023-11-29:
@Slyde-R 

- If we opt for a 4-pin connector, the ArduinoMicroShieldBoard01 PCB will have to be adapted. This shouldn't be a problem as there is more than enough space as you can see in the screenshot.  
[image]

- The dimensions of the replacement headphone jack should be fairly similar, but may vary slightly as there is quite a lot of space in the design. (This should be checked again once a headphone jack has been selected and the PCB has been adapted).  
[image]

- Furthermore, a different (4pin) audio cable needs to be selected in order to ensure compatibly with the used headphone jack. (already marked in red in BOM).

Slyde-R, 2023-12-03:
edited PCB files to accomodate different 3pin jack exported files can be found [here](https://drive.google.com/drive/folders/1maboma9Yz2E8RrRPJgDyxDnrLxclWt5I?usp=drive_link).
The new subtotal is because of the lead free finish:

[image]

This is with the most expensive and also quickest shipping as the slowest takes 14-16 business days and other slower ones only save 1.98EUR

Slyde-R, 2023-12-03:
-> ready to order

closed 2023-12-03 by Slyde-R

Slyde-R, 2023-12-11:
I ordeered the PCBs

[invoice.pdf](https://github.com/CyberducksTeam/HACKberry/files/13641230/invoice.pdf)