CyberducksTeam/HACKberry #3: [Networking] Connect with HACKBERRY

sinahsoehner, 2023-11-17:


sinahsoehner, 2023-11-18:
Message sent via contact formular on the HACKberry homepage.

Message:
Dear HACKberry Team,

I hope this message finds you well. 
As a non profit organization in Linz, Austria we represent Team CYBERDUCKS (more information you can find on our homepage: https://cyberducks.at/), a group of individuals passionate about making a positive impact through medical technology. We recently came across your nonprofit and were truly inspired by the incredible open source projects.

In particular, our next step is the arm prosthesis project outlined on your website.  As a team with a shared commitment to making a difference, we would like to express our interest in collaborating with your organization to contribute to the development of these prosthetics.

Our team has experience in medical engineering topics and projects, and we believe that our skills align well with the goals of your arm prosthesis project. We are eager to lend our expertise to enhance the project's capabilities and contribute to the overall success of your mission.

Before we start the ARM project, we were wondering if you could provide any additional information or specific requirements for the arm prosthesis project. Specifically, if there are any tips or best practices based on your experience with similar projects, we would be grateful to receive them. We want to make sure that our project and teamwork is as effective and meaningful as possible.

We are already excited about response and the ARM project!

Kind regards,
Team CYBERDUCKS

Slyde-R, 2023-11-20:
👌 
Bin jetzt schon ein Fan von dem Projektmanagement^^

Ich würde vorschlagen, dass du das issue schon closen kannst und dann vllt nach "In Review" reinschieben, damit man die eventuelle Antwort auch noch einfügen kann

@sinahsoehner

Slyde-R, 2023-12-01:
Answer:

Dear Team CYBERDUCKS,

I am Genta, the founder of the HACKberry project, and I'm sorry for my late reply.
You may not have noticed but HACKberry was exhibited at Ars Electronica back in 2016. So your message has made me feel happy.

Now HACKberry is managed solely by myself and the project hasn't been active as before. However, I am still selling components of HACKberry overseas and replying to any specific inquiries, both from technical and social aspects.

I can answer if you give me a more specific questions list.
Or, If you want to start more casually, I can set up an online meeting with your team.

Best,
Genta  

-- 
====
近藤玄大 Genta Kondo
特定非営利活動法人Mission ARM Japan 
[genta.kondo@mission-arm.jp](mailto:genta.kondo@mission-arm.jp)

closed 2023-12-01 by Slyde-R